Livenow14/2020_INU_Embedded_System #3: 03 Odroid& Hosts 환경설정 13p _ No such file or directory

Comely-U, 2020-04-07:
[image]
 
hello.c 파일로 toolchain test 과정 중 ./hello_arm을 했을 때
bash: ./hello_arm: cannot excute binary file: Exec format error 가 아닌
/lib/ld-linux-armhf.so.3: No such file or directory 이 출력됩니다.
이 경우 어떻게 해결해야 하나요?

사용환경은 MacOs 카탈리나 + Parallels에 Ubuntu 18.04 64bit입니다.

Livenow14, 2020-04-08:
1. arm-linux-gnueabihf-gcc로 다시 컴파일을 해주세요. 
ppt에 나와있듯이 sudo apt-get install libc6-i386 을 실행후 다시 컴파일을 해주시면 오류가 해결될겁니다.  (이미 컴파일된 실행파일을 실행하는 것이아닌 재컴파일을 말함.)

2. 위의 문제가 해결이 안될 시 PATH  설정이 제대로 안되있을 가능성이 높습니다. 
 No such file or directory  오류는 PATH설정이 안되있을 때 자주 나타나는 오류이기 때문에, PATH설정을 다시한번 해보시길 바랍니다. 

3.위와 같은 방법이 모두 통하지 않을때
->  지금 실행을 못시켜도 이후 minicom을 이용하여 odroid 보드 내의 server에 진입하여 server내의 환경에서 gcc옵션을 통해 실행파일을 만들 수 있습니다. (저또한 이렇게 컴파일을 하였습니다. ) 
-> 지금 하는 수업의 취지는 odroid server내의 vim을 이용하여 코드를 작성하면  가독성과 코드 정리가 제대로 되지 않을 가능성이 있기 때문에, minicom을 이용한 코드 컴파일이 아닌 host에서 편하게 코드를 작성하자는 취지입니다.  ( ubuntu에서는 여러 코드 작성기를 지원합니다. 하지만 odroid보드 내에서는 지원을 하지않습니다.  ex) gvim ) 
host 에서 arm용 실행파일을 만들 수 있다는 점은 굉장히 편리합니다. 
-> 여기서 죄송하다고 말씀드릴 수 있는 점은, odroid 사양은 모든 학생이 동일하나 그것을 실행하는 host의 환경이 각각 달라 저또한 오류 분석이 쉽지 않은점입니다. 
1,2번으로 해결이 안될시에는 개별적으로 원격을 통해 해결하거나 실시간으로 오류를 잡아드리겠습니다. comment를 달아주세요

Comely-U, 2020-04-08:
[image]
1번과 2번으로도 해결이 되지 않습니다.
sudo apt-get install libc6-i386을 한 후 다시 컴파일 한 파일도 똑같은 오류 메시지가 떴고,  arm에 tab키 두번을 눌러서 명령어들을 확인할 수 있었기에 PATH 설정 문제는 아니었던 것 같습니다.
우선,  file 명령어를 통해 hello_arm의 정보에서 오류메시지로 나온 /lib/ld-linux-armhf.so.3 이 아닌 /lib/ld- 로만 나와있는 것을 확인하였고, /lib 디렉토리 안에 ld-linux-armhf.so.3과 ld-linux-arm.so.3 이 없는 것을 확인하였습니다. 이 경우 해결 방법이 있을까요?

Livenow14, 2020-04-08:
조금더 확인해보니까 강의대로 진행한다면 ld-linux.so.2가 생기네요. 
ld-linux-armhf.so.3.이 없다는 오류가 뜨는건 파일이 없어서 생기는 오류인 것 같아요( 저도 처음겪어보는 현상이라 찾아봤습니다 ) 
밑의 명령을 실행 해보시고 다시 컴파일 해보시겠어요 ? 
sudo ln -s  /lib/ld-linux.so.2 /lib/ld-linux-armhf.so.3
(심볼릭 링크를통해 프로그램이 메모리에 적재되는 시점에서 실행에 필요한 라이브러리를 링킹 해줌)

Comely-U, 2020-04-08:
[image]
/lib/ld-linux-armhf.so.3: Invalid ELF image for this architecture 이 뜨는 것으로 보아 해결된 것 같습니다. 감사합니다!