Livenow14/2020_INU_Embedded_System #4: 03 Odroid& Hosts 환경설정 13p_arm용 gcc로 컴파일하기

GHeeJeon, 2020-04-12:
[image]
arm용 gcc로 컴파일이 안됩니다.
arm-linux-gnueabihf-gcc: error trying to exec 'cc1': execvp: No such file or directory 가 출력됩니다.

구글링하여 해결법을 찾아보았지만,
1)sudo apt-get update
sudo apt-get install --reinstall build-essential

2)sudo apt-get install g++

3)sudo apt-get install gcc

여전히 컴파일이 되지 않습니다. 어떤 문제가 있는 걸까요...?
Vm Ware_ Linux Ubuntu 18.04.4를 사용하고 있습니다.

Livenow14, 2020-04-12:
1. ppt의 설치과정을 다 하셨나요?
 - 어느 한곳이 빠져있는지 다시한번 강의의 내용그대로 진행해 보시길 바랍니다. 
 - sudo apt-get install libc6-i386을 실행 후 다시 컴파일 해보시길 바랍니다. 

2. path 적용이 안되어있을 때 
 - path 설정이 정확하게 되지 않아 발생하는 문제일 수 있습니다 . path설정 후 arm-linux- 후 tab을 치면 나오는지 확인 바랍니다. 

3. 1, 2번이 안될시,
- 대다수의 오류는 어떠한 과정을 생략했을 때 발생합니다. 그 과정이 어떠한질 모르기 때문에 오류 fix가 쉽지않은점 양해바랍니다.
- 커맨드를 남겨주시면 팀뷰어 원격 조종을 통해 학생의 컴퓨터에 접근 후, 오류 fix를 해보도록 하갰습니다.

GHeeJeon, 2020-04-12:
1. 다시 설치해보았으나, 여전히 컴파일이 되지 않습니다.
2. path설정 후 arm-linux- 후 tab을 쳤을 경우, arm-linux-로 시작하는 여러 명령어가 출력됩니다.
3. 팀뷰어 원격 조종을 통한 오류확인을 부탁드리고 싶은데, 언제가 편하신가요?

Livenow14, 2020-04-13:
네 따로 연락드리겠습니다. 
이러닝의 메세지를 확인해주세요.

closed 2022-08-11 by GHeeJeon