Livenow14/2020_INU_Embedded_System #11: 오드로이드 초기 설정

yd4825, 2020-04-23:
오늘 이러닝 메시지로 메시지를 받았는데 이해가 잘 안가서 여쭤보게 되었습니다.

1. https://odroid.in/ubuntu_14.04lts/ 에 들어가, 14.04.3lts 중 용량이 제일큰것을 받아주세요 
2. https://www.balena.io/etcher/ 에서 학생의 윈도우 버전에 맞는 것을 선택해서 다운받아주세요
3. odroid - c1에 꽃혀있는 micro-sd카드를 리더기를 통해 컴퓨터와 연결한 후 
4. etcher로 부트 이미지(다운 받았던 14.04.3)를 삽입해주세요 
5. 이후 odroid 환경설정을 처음부터 진행해주세요 

1번에 14.04.3lts 중 용량이 제일큰  1.1G를 리눅스에서 받으면 되는건가요?
4번에 이미지 삽입을 어떻게 하는지 궁금합니다.
5번 까지 마치면 우분투를 삭제하고 다시 설치하여야 하나요?

1번 2번이 설치 속도가 느려 10시간이 넘게 걸리게 되어 확인할 수 없어서 물어 보게 되었습니다.
죄송합니다.

Livenow14, 2020-04-23:
1. 아니요, 리눅스(vm)에서 받는것이 아닌  학생의 윈도우에서 진행해주세요 
2. etcher 사용법을 치면 자세히 나와있습니다. 부트이미지 삽입할 디스크 선택, 이미지 선택, flash 가 끝입니다. 
3. 이것은 vmware의 리눅스를 재설치하는 과정이아닌,  odroid-c1의 micro-sd카드에 들어있는 odriod용 리눅스 파일을 재 설치하는 과정입니다.

yd4825, 2020-04-23:
정말 죄송하지만 
1번에서 클릭을 누르면 파일이 깨져서 다운을 받지 못하겠습니다.
이렇게 되는데 어떻게 하면 될가요?