Livenow14/2020_INU_Embedded_System #12: 아까 질문을 삭제해서 다시 여쭤봅니다.

yd4825, 2020-04-23:
1. https://odroid.in/ubuntu_14.04lts/ 에 들어가, 14.04.3lts 중 용량이 제일큰것을 받아주세요 
2. https://www.balena.io/etcher/ 에서 학생의 윈도우 버전에 맞는 것을 선택해서 다운받아주세요
3. odroid - c1에 꽃혀있는 micro-sd카드를 리더기를 통해 컴퓨터와 연결한 후 
4. etcher로 부트 이미지(다운 받았던 14.04.3)를 삽입해주세요 
5. 이후 odroid 환경설정을 처음부터 진행해주세요 

1번이 클릭을 하면 글자가 다깨진 창이 나옵니다.
어떻게 다운을 받는지 모르겠어서 여쭤봅니다.
가상머신에서는 클릭을 하면 다운이 되는데
윈도우에서는 클릭을 하면 글자가 깨진 화면이 나옵니다.

yd4825, 2020-04-23:
넵 한번해보겠습니다.

yd4825, 2020-04-23:
다운로드 속도가 너무 느려 20시간이 넘게 걸리는데 
그래도 기다려서 받아보는게 좋을까요?

yd4825, 2020-04-23:
다운로드 시간이 20시간이 넘어가고 도중에 다운로드가 중단되어 다시 실행하면 작업중단이되어서 처음부터 다시 다운로드를 해야됩니다. 혹시 sd카드를 택배로 보내도 될까요? 하루종일 해결이 안되서 답답해서 그런거 같습니다. 죄송합니다.

Livenow14, 2020-04-23:
집의 인터넷이 느리지 않고서 그렇게 오래 걸리지않습니다. 
1기가 바이트 밖에 안되는 용량이 20시간 이상 걸린다면 어떠한 문제가 있는 것입니다. 
하드커널사의 서버문제일 수 있으니 아래 링크에서 다운받아보세요 

https://dn.odroid.com/S805/Ubuntu/

택배는 받지 못합니다..

yd4825, 2020-04-23:
인터넷이 느리편은 아니지만 잘모르겠습니다.
한번 다시 다운받아보겠습니다.

Livenow14, 2020-04-23:
이러닝 메세지 확인해주세요

yd4825, 2020-04-23:
메시지 확인했습니다!