Livenow14/2020_INU_Embedded_System #13: 오드로이드 설정

jhc324, 2020-04-24:
[image]

마운트 하는동안 오류가 발생하였는데 어떻게 해결해야하나요?  Fusing 까지 했었는데 GIPO 강의 들으면서 전원연결을 하였더니 문제가 발생했습니다.

Livenow14, 2020-04-25:
As mentioned during the first class, all odroids are in a fully checked state.
But problems arise for a variety of reasons. Let's check.

First, just wait for a 10 minutes. 
Odroid File System will check automatically for problems.

Second, I don't know in detail about your problem. 
If the first solution didn't work, I need to scan your computer through remote control(Team- Viewer).
Please tell me your class(daytime / nighttime), and name using Github Comment.
I will connect you in e-learning message.

jhc324, 2020-04-26:
nighttime, choi ji han

jhc324, 2020-04-26:
at nighttime , name : choi ji han