Livenow14/2020_INU_Embedded_System #14: 03 Odroid Host 환경설정 p29 hello파일이 없습니다.

wonyyy, 2020-04-29:
[image]

zmodem을 누르고 들어갔는데 hello 파일이 뜨지 않습니다. ㅠ
아래사진처럼 파일생성을 다했는데 ordroid내에서 찾을 수 없습니다.
[image]

Livenow14, 2020-04-30:
1. zmodem에서 폴더를 이동해 봤나요??? 
zmodem안에서도 폴더를 이동해 볼 수 있습니다. 확인해주세요.

2. zmodem에 파일이 뜨지않는경우는 host와 연결이 제대로 되지 않았을 때 발생합니다. 
odriod configuration에 usbtty가 제대로 설정 되어있는지 확인해주세요 

3. 위와 같은 해결방법으로 해결이 안될 시 원격을 통해 fix하도록 하겠습니다.
주/야간 반과 성함을 알려주시면 이러닝 메세지로 연락드리겠습니다.

wonyyy, 2020-04-30:
다른위치에 우분투를 설치해서 그런가봅니다 ㅠ c드라이브에 다시 설치해서 해결했습니다!! 감사합니다