Livenow14/2020_INU_Embedded_System #21: 05-3 CLCD lab 6p_버튼 사용에 관한 질문

minj-shalom, 2020-05-12:
버튼 사용에 대해 수업시간에 구체적으로 다루지 않아 중간고사를 하기 위해 지금 먼저 버튼에 관한 테스트를 진행하고 있습니다. 버튼값을 입력받아 버튼의 상태에 따라 led를 출력하는 간단한 테스트를 진행중인데요. 보시는것처럼 빵판에 설계했고, 이렇게 코드를 짜봤습니다. 버튼을 누르면 led가 켜져야 하는데, 컴파일 후 프로그램을 실행시키면 led가 버튼을 누르는 것과 별개로 계속 출력되는 현상이 발생합니다. 하드웨어 설계나 코딩 과정에서 잘못된 부분이 있나요?
[image]
[image]

Livenow14, 2020-05-13:
수업에서 설명햇듯이, 빵판의 좌우는 영향을 미치지않습니다. 
다시한번 그 부분에 대한 강의를 수강하시길 바랍니다.

minj-shalom, 2020-05-13:
보시는 것처럼 중간고사 direction에 나와있는 그대로 하드웨어 세팅을 다시 하고 실험을 진행해봤으나 여전히 버튼 입력이 되지 않습니다. 연결되어 있는 한 라인에 같은 부품을 꽂을 경우 내부에서 합선이 일어나 동작하지 않는다는 빵판의 원리는 충분히 숙지하고 있습니다. 보시는 사진 외에도 버튼의 방향이나 저항의 연결 유무 등 다양한 조건을 고려하여 여러 실험을 진행해봤으나 여전히 버튼값을 입력할 수 없었습니다. 왜그런걸까요?
[image]

Livenow14, 2020-05-13:
위의 코드상으로 봤을때, digitalRead부분이 잘못되있는 것 같습니다. 
중간고사 direction에서 어떻게 사용하는지 다시한번 봐주세요.

minj-shalom, 2020-05-13:
중간고사 direction에는 digitalRead 함수에 대한 구체적인 설명이 나와있지 않습니다. 보시는 사진은 중간고사 direction ppt의 5번 슬라이드입니다. 여기에 digitalRead에 대해 아주 간략한 설명만 나와있고, 그 외에는 digitalRead에 대한 언급이 없습니다. 그래서 보시는 사진처럼 구글링을 통해 digitalRead 함수에 대한 구체적인 정보와 사용 예시를 얻어서 다음과 같이 코드 수정 후 실험을 진행해봤으나 여전히 해결되지 않습니다. 보시는 코드 외에도 코드를 조금씩 수정해가며 여러 코드로 실험을 해봤으나 마찬가지입니다.
이번 lab에서 GPIO의 버튼은 입력을 받는 기초적이면서 중요한 요소임에도 불구하고 수업 시간에 이 부분에 대해서 구체적으로 다룬 적이 없고, 계속해서 많은 학생들이 이 부분에 대해 질문이 이어지는 것 같습니다. 따라서 버튼 사용과 관련해서 추가 설명이 필요해보이는데 이 부분에 대한 강의 영상이 계획되어 있지는 않은지 궁금합니다. 또한 하드웨어 설계나 코드 상에서 별다른 문제 사항이 없음에도 왜 이런 문제가 발생하는지 답변해주시면 감사하겠습니다.
[image]
[image]
[image]

Livenow14, 2020-05-13:
[image]

1.  Direction ppt의 hint의 뒤 부분을 참고하면,  pullUpDown의 방식을 PUD_DOWN, PUD_ UP으로 한것을 볼 수 있습니다. PUD_UP이 먹히지 않다면, PUD_DOWN을 사용하세요. 

2. 이후 버튼입력에 대한 추가적인 동영상은 제작되지 않습니다. 
3학년 전공수업인 만큼 이해할 수 있는 선에서 ppt와 강의를 제작하였습니다.  
써야하는 함수, 그 함수에 대한 설명,  함수가 반환하는 값, 하드웨어 상 버튼의 pin배열 방식 등, 강의와 ppt에서 설명하였습니다.
그 이상의 direction은 과제에 직접적인 hint가 될 수 있어 설명하지 않았습니다. 이론 수업이 아닌 실습 수업이기 때문이고, hint가 곧 성적에 반영되기 때문입니다.  
학생과 반대로 많은 hint를 주는것에 대한 불만을 가지고 연락을 하는 학생 또한 있습니다. 

3. 피드백을 어느정도 까지 줄 수 있는지에 대한 모호함으로 앞서 우회적으로 힌트를 드렸습니다.
인터넷으로 진행되는 실습 수업이기 때문에, 부족함이 많습니다.
이후 진행될 강의에서는 학생의 피드백을 반영하겠습니다. 감사합니다.