Livenow14/2020_INU_Embedded_System #23: <중간고사>WiringPi GPIO관련 질문 드립니다.

JOUNHO, 2020-05-15:
[image]
위에 사진은 프로젝트 진행중 wiringPi num을 define 해준 것입니다.
프로젝트를 진행도중에 NUM1에 해당하는 wiringPi num 22(label로는 #100)가 문제가 있다는 것을 알았습니다. 1을 제외한 0~9가 모두 clcd에 출력도 되고 감가산기가 정상 작동 되어지는데 wiringPi num 22(숫자 1)만 작동이 되지 않고 있습니다.
 점퍼선을 바꿔보고 스위치도 바꿔도 보았지만 문제는 해결되지 않았습니다.
또한 num2에 wiringPi num 22를 주고 num1에 wiringPi  21로 바꿔서 실행 했을땐 숫자 1이 출력되고 2만 출력이 되지 않았습니다.  그래서 blink파일을 이용하여 wiringPi num만 바꿔주면서 LED를 켜보았는데 마찬가지로 wiringPi num 22일때만 LED의 불이 켜지지 않았습니다.
이 문제를 어떻게 해결해야할지 몰라서 질문 드렸습니다. 감사합니다.

Livenow14, 2020-05-16:
pinNum를 다르게 해주세요. 
22번이 안될 수 있습니다.

JOUNHO, 2020-05-16:
중간고사 프로젝트에 pinNum의 개수가 LCD6개 숫자10개 +,-,= 3개로 총 19개를 사용해야하는데 저희가 사용할 수 있는 pin의 개수가 19개인걸로 알고 있는데 19개보다 더 사용 할 수 있는 건가요??

Livenow14, 2020-05-17:
네, 사용할 수 있습니다. 
다만  과제 채점시 그 방법이 안먹힐 수 있습니다.

학생의 하드웨어에서만 안되는 경우 일 수 있으니
우선, 되지 않는 번호는 놔두고 로직은 그대로 구현해주세요 
ex) 22번의 pinNum에 물린 버튼이 먹히지 않지만, 21번으로 가능하니 21번으로 버튼을 세팅한 후 로직이 문제없음을 확인하고, 과제 제출시에 22번으로 바꾸어서 제출.

JOUNHO, 2020-05-17:
넵 알겠습니다. 답변해주셔서 감사합니다.