Livenow14/2020_INU_Embedded_System #39: TFT 과제 관련하여 여쭤봅니다

EUNJI1012, 2020-06-02:
안녕하세요! ppt 3쪽에 적혀있는 화면 정보와 관련해서 여쭤봅니다

LCD가 320x240 pixel인데 세로를 계산해보면

 앞 글자 8줄 + 라인 사이 7칸
8 lines x 24 pixel + 7 lines x 8 pixel = 192 + 56 = 248 > 240
이 나옵니다.

혹시 제가 틀린 것이 있는지 확인 부탁드립니다. 감사합니다!

Livenow14, 2020-06-03:
1. 라인이 화면을 넘어가는 경우에서, 오류헨들링을 해주시면 됩니다. 
(ex. 짤린채로 입력, or 전체 화면을 위로 한칸 이동, 단 보드가 멈추지 않게)

2. 문자를 출력할때는, 그 방법이 다양합니다. 
글자를 출력하는 방법에 대해 언급하지 않은점은, 어떠한 방법을 써도 출력이 됨을 확인하면 인정하겠다는이야기 입니다. 감사합니다.

minj-shalom, 2020-06-06:
1번 질문과 관련해서 추가 질문 남깁니다.
픽셀을 계산해보면 direction에 나와있는대로 8줄을 전부 출력할 경우 세로줄이 잘리는 현상이 발생합니다. 이 경우 잘린채로 입력하거나 전체 화면을 위로 한 칸 이동하는 등의 error handling을 하면 된다고 하셨는데, 7줄까지만 출력하고 8줄이 넘어갈 경우 overflow라는 문구를 출력하는 방식으로 error handling을 해도 될까요?

Livenow14, 2020-06-06:
error handling은 자유입니다. 
다만, 보드가 멈추거나, 화면이 멈추는 등의 error handling은 error handle을 하지 않았다고 판단합니다.