Livenow14/2020_INU_Embedded_System #42: C 컴파일 질문

gopherclass, 2020-06-08:
오드로이드가 사용하고 있는 컴파일러가 낡아 C 표준을 따르도록 코드를 작성해도 컴파일 옵션을 일일이 지정하지 않으면 기본적으로 20년전 표준을 사용하고 있는 것 같아 컴파일이 되지 않습니다. 직접 컴파일 옵션을 제어하면 어느정도 해결이 예상되는 문제이나 일부 문제는 그렇지 않을 수 있고 또한 생각이 다르실 수 있으므로 질문드립니다.

컴파일할 때 사용하는 C 표준은 어떤 버전을 따라야 하나요? 학생마다 표준을 완벽히 따르지만 다른 C 표준을 따르고 있는 경우에는 어떻게 하나요? 작성하는 C 코드는 의도하는 표준을 얼마나 따르고 있어야 하나요? 표준을 따르는 수준을 제어하거나 에러 수준을 제어하는 컴파일 옵션에 따라 컴파일이 될수도 안될수도 있는데 어떻게 해야 하나요? 에러 수준을 제어하지만 -Wwrite-strings 같이 extra 에도 속하지 않은 에러 수준을 제어하는 옵션들이 포함될 수 있다고 가정해야 하나요? C 언어 명세 표준은 아니지만 보드에서 지원하는 코드를 사용할 수 있나요? 소스 코드가 여러 translation unit으로 나누어져 있다면 학생이 직접 컴파일 옵션을 관리할 수 있어야 할텐데 어떻게 하면 되나요?

과제에 사용되는 C 코드를 만드는 코드는 제출해야 하나요? 이 코드는 C로 작성되어 있지 않지만 최종적으로 컴파일할 코드가 준비되어 있고 그 코드가 C면 되는 건가요?

위 질문들은 C에 국한시켜 질문하고 있지만 C와 C++은 이젠 완전히 서로 갈라진 별개의 언어이기 때문에 C++의 컴파일 옵션은 C와 의미적으로 전혀 다를수 있습니다. C와 C++의 표준이 완전히 다르기 때문에 위 질문들을 C++로 바꿔봤을 땐 어떻게 하면 되나요?

Livenow14, 2020-06-08:
수업에 쓰이는 버전과 표준은 교수님이 결정하실 사항이라 생각하여 
제가 이렇다 할 판단을 내릴 수 없는 질문이라 생각합니다. 

교수님 이메일인 hbbaek359@gmail.com 로 질문하여 주세요. 감사합니다.

GHeeJeon, 2020-06-12:
저와 같은 고민을 하시는 것 같아 답변 남깁니다. 저는 `Makefile`에서 컴파일 옵션으로 `-std=c99 -Wall` 주었습니다.
GCC extension 사용하고자 하신다면 `-std=gnu99` 쓰시면 더 좋아요!