Livenow14/2020_INU_Embedded_System #44: 공백 문자 출력 질문

gopherclass, 2020-06-08:
`hello. this is an example. capital letter cannot be typed.` 을 입력하기 위해서는 공백 문자가 필요해보입니다. 하지만 어디에도 공백 문자를 입력하는 방법에 대한 명시적인 설명이 없습니다. 짐작가는 바로 커서 이동을 하면 자동으로 공백 문자가 입력되게끔도 해볼 수 있으나 이 방법은 공백을 문자 끝에 추가할 수만 있고 문자열 중간에 공백을 삽입할 수 없다는 문제가 있습니다. 그렇다면 그냥 공백 문자 없이 모든 문자를 붙여 써서 `hello.thisisanexample.capitallettercannotbetyped.`라고 입력하면 됩니까?

추가로 `입력 화면은 스크린 가운데 위치` 문구가 무슨 의미인지 모르겠습니다. 마치 왼쪽 정렬한 듯이 한 줄당 10 자씩 출력하면 안되고 마치 가운데 정렬한 듯이 출력해야 된다는 의미입니까?

WhiteHyun, 2020-06-08:
[image]

지나가는 학생입니다.
다른 학생도 이와 동일한 질문을 했었는데 교수님께서 무시하고 진행해도 된다고 답변했었습니다.
사진 참고 하시면 되겠습니다.

Livenow14, 2020-06-08:
공백이 포함된 문자를 출력하라고 명시해 뒀으니
본인의 방법대로, direction에 어긋나지 않게 공백을 출력하게 해주세요.
 그 방법에 대한 설명은, 리포트에 적어 제출하면 되겠습니다.