Livenow14/2020_INU_Embedded_System #49: gpio pin 입력 질문입니다.

Goldvista, 2020-06-11:
interrupt를 이용하여 버튼 상태 변화를 감지하여 입력받도록 프로그램을 작성하였는데, 프로그램을 실행하면 지정해준 118번 pin과 관련하여 오류가 납니다.
[image]
lcd에 미리 할당되어 있는 핀이라 오류가 발생하는것 같은데, 어떻게 해야 할까요?

Goldvista, 2020-06-12:
polling 방식으로 바꾸어 봐도 118번은 입력이 되질 않습니다.

Livenow14, 2020-06-12:
ppt에서 118번은 LCD용으로 사용하지 않는다고 명시해두었습니다. 
다른핀을 사용해주세요 
감사합니다.

Goldvista, 2020-06-12:
[image]
ppt에서 wxy 버튼을 wiringpi 11번에 연결하라고 되어 있는데, 11번이 118번 #118입니다. 제가 잘못 이해한건가요?

Livenow14, 2020-06-14:
wiringPi 11번이 Lable 118번이었네요, 
핀에 오류가 생긴다면, 다른핀에 하시고, 리포트에 문제가 있어 변경하였다는 문구 남겨주세요.