Livenow14/2020_INU_Embedded_System #55: 07 Touch screen (all in one) 5p_Touch screen error

minj-shalom, 2020-06-17:
아래 보시는 사진과 같이 Touch screen에 필요한 device driver를 모두 설치했고 event4 handler를 정상적으로 확인했습니다. 그리고 강의자료 5번 slide에 있는 code를 제대로 옮겨서 정상적으로 compile까지 마쳤습니다. 그러나 실행을 시켜보면 이렇게 바로 program이 종료됩니다. 왜그런가요?

[image]
[image]
[image]
[image]

minj-shalom, 2020-06-17:
event는 기기에 따라 1이나 4가 될 수 있다는 comment가 있어서 1도 해봤으나 결과는 마찬가지입니다.

potados99, 2020-06-17:
지나가는 학생입니다.

바로 프로그램이 종료된다고 하셨습니다.

주어진 코드에서 에러 출력(ex. Segmentation fault)도 없이 바로 프로그램이 종료되는 경우는 `fd`가 0보다 작아 while 루프에 진입하지 않고 바로 `-1`을 반환하며 프로그램을 종료하는 경우 밖에 없는 것으로 보입니다.

`fd`가 0보다 작은 경우는 파일 열기에 실패한 경우입니다. `man`에 의하면 `open()`은 오류가 발생하였을 때에 `-1`을 반환합니다.

>  open(), openat(), and creat()  return  the new  file  descriptor,  or  -1 if an error occurred (in  which  case,  errno  is  set appropriately).

자세한 이유는 `errno`를 확인하여야 알 수 있지만, 주어진 코드로 보아 `open()`의 첫 번째 인자에 문제가 있는 것으로 보입니다.

`open()`에 넘겨주신 인자 `dev/input/event4`는 상대 경로입니다. Slash로 시작하는 절대 경로로 `/dev/input/event4`와 같이 넘겨주신다면 성공하지 않을까 싶습니다.

minj-shalom, 2020-06-18:
아아 저걸 못봤네요 ㅜㅜ 감사합니다!

Livenow14, 2020-06-18:
정확한 답변, 감사드립니다.

minj-shalom, 2020-06-18:
direction에 주어진 code를 조금 수정해서 간단한 calibration test를 진행중입니다. user가 screen을 touch하면 해당 좌표를 출력하는 code인데요. 여기에 영상을 올릴 수가 없어서 못보여드리지만 생각보다 touch 반응 속도가 좀 느리기도 하고 calibration 결과가 약간 부정확한 것 같습니다. 특별히 coding 과정에서 복잡하게 coding한 부분도 없고, 주어진 code를 조금 수정한 정도라 원인을 잘 모르겠습니다. 원래 이런건가요? 영상을 보여드리면 좋을거 같은데 아쉽군요.

Livenow14, 2020-06-19:
원래 calibration과정은 과제에 포함된 부분이었습니다. 
교수님께서 학생들이 쉽게 과제를 할 수 없을것이라 생각하여 calibration 코드를 공개한 것입니다.
calibration 결과가 정확하지 않다고 판단될 경우, 학생 스스로 calibration의 코드를 작성하고 비교해보시길 바랍니다. 

또한 속도 부분은 코드의 효율성이 큰 부분을 차지하며, 이는 본인의 코드 디자인에 따라 달라집니다. 
gcc의 옵션에 빨라지게 하는 옵션이 있으나, 연산을 일정 이상 무시해버리는 경향이 있기때문에, 적절히 사용하시길 바랍니다.

minj-shalom, 2020-06-19:
https://youtu.be/HpqMhgoOQQk

GitHub에 영상을 올릴 수가 없어서 Youtube에 영상을 올렸습니다. 앞서 말씀드린 Touch Screen Calibration Test 영상입니다. 이정도의 정확도와 속도인데 괜찮을까요?

Livenow14, 2020-06-19:
이정도 속도는 아주 괜찮습니다.